openscopeproject/InteractiveHtmlBom #228: generated bom html is empty

rpolasek, 2021-02-25:
when i generate a bom and open it in the chromium the bom is emtpy. i'm using last version from github and python v3.9.1. i'm enclosing the generated bom html (zip) and also a screen-shot of my browser with developer tools (jpg) opened.

[dramduino.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/6045460/dramduino.zip)

[image]

qu1ck, 2021-02-25:
Most likely you didn't restart kicad after updating plugin or didn't remove previous ibom files before updating.

rpolasek, 2021-02-25:
@qu1ck you're right that was the problem. thank you.

closed 2021-02-25 by qu1ck