openscopeproject/InteractiveHtmlBom #229: Support curved tracks

Funkenjaeger, 2021-02-27:
KiCad may not support curved tracks, but other EDA tools do (e.g. Eagle/Fusion 360 Electronics via GenericJSON).  
Recommend adding optional fields to the 'tracks' object in the pcbdata struct to support rendering curved traces.

qu1ck, 2021-02-27:
Kicad also supports them in nightly. I'll add this soon.

Funkenjaeger, 2021-02-27:
I've got this working on my fork with Eagle/Fusion.  I assumed the same structure as the 'arc' drawing object (start, radius, startangle, endangle).  The changes to render.js to make that work turned out to be trivial (which was good, since I'm not versed in javascript)

Do you know enough about the Kicad changes to define the revised format for tracks?  If so, I can try to conform to that and submit a PR for the update to render.js and the genericJSON schema (I'll leave the actual Kicad parser code to you)
[image]

qu1ck, 2021-02-28:
Yes, arcs will be defined exactly like you describe. They are not accessible from python yet but I sent a MR for that
https://gitlab.com/kicad/code/kicad/-/merge_requests/701

Funkenjaeger, 2021-02-28:
Submitted PR #230

commit af0a57e2 (2021-03-01): Support arc tracks in kicad

Fixes #229
closed 2021-03-01 by qu1ck