openscopeproject/InteractiveHtmlBom #231: Do not mount parts

leoheck, 2021-03-01:
My board has DNM (do not mount) components tagged on the Schematic. 
Do we have any way to avoid these components when creating the interactive BOM?

iharshadec, 2021-03-01:
This is interesting. I used to place DNP - Do not Place lable and yes if we get some indication that would be really great.

leoheck, 2021-03-01:
Yeah, a field with a custom label would be nice, so you could use the DNP and I DNM. 
I have this custom BOM script that generates a simple bom.csv.
This one does not include components that do not have a footprint or MPN (manufacture part number). It also filters out components with DNM tags.
So, if we could have something that does the same for the Interactive BOM, it would be awesome.

qu1ck, 2021-03-01:
[image]

leoheck, 2021-03-02:
Oh, @qu1ck thank you so much, I am dumb.

closed 2021-03-02 by leoheck