openscopeproject/InteractiveHtmlBom #236: KiCad 5.99 compatibility

lienbacher, 2021-03-12:
As KiCad 6 is rapidly maturing, are you planning on updating this to run with 5.99? 

I just tried it and it failed:

```
Traceback (most recent call last):

File "/Users/user/Documents/KiCad/5.99/scripting/plugins/InteractiveHtmlBom/ecad/kicad.py", line 523, in Run
ibom.run_with_dialog(parser, config, logger)

File "/Users/user/Documents/KiCad/5.99/scripting/plugins/InteractiveHtmlBom/core/ibom.py", line 328, in run_with_dialog
main(parser, config, logger)
```

qu1ck, 2021-03-12:
Use latest version from master branch, not release archive. Also make sure your nightly is up to date. This is addressed in the FAQ.

I tried with current nightly and ibom works fine for me, if you still have an issue after update then post full error message. For python the meat of the exception is at the bottom, unlike java.

lienbacher, 2021-03-12:
dang, right! that did the trick, sorry for the hassle!

closed 2021-03-12 by lienbacher