openscopeproject/InteractiveHtmlBom #238: wx._core.PyNoAppError: The wx.App object must be created first

reinauer, 2021-03-15:
[image]

I get this error message upon trying to start the plugin.

qu1ck, 2021-03-15:
Please share full kicad version from about dialog (press "copy version information" button).

reinauer, 2021-03-15:
Application: KiCad
Version: 5.1.9-73d0e3b20d~88~ubuntu20.10.1, release build
Libraries:
    wxWidgets 3.0.5
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.9 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh/0.9.3/openssl/zlib nghttp2/1.41.0 librtmp/2.3
Platform: Linux 5.8.0-44-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.68.0
    Compiler: GCC 10.2.0 with C++ ABI 1014
Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

qu1ck, 2021-03-15:
That is strange. Do any other plugins work for you?
Try this one:
[dummy_plugin_1.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/6139392/dummy_plugin_1.zip)

reinauer, 2021-03-15:
This one does not seem to show up in the external plugins menu under Tools, so I'm not sure.

I have a snapeda plugin installed that works fine, if that's any reference. I can try some other plugins later

qu1ck, 2021-03-15:
Make sure when you extracted it there is only one folder (i.e. not nested `dummy_plugin_1/dummy_plugin1/...`)

Snapeda plugin doesn't use wx for gui, it uses tk so it working is not an indicator.
You can try some of these plugins: https://github.com/MitjaNemec/Kicad_action_plugins

qu1ck, 2021-03-15:
Oh, actually I screwed up that dummy plugin, here is the correct version
[dummy_plugin_1.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/6144370/dummy_plugin_1.zip)

reinauer, 2021-03-16:
Hm. This one dies in wx.MessageBox("Plugin 1") with the same error message (wx.App object must be created first).

As an additional data point, running the script on the command line works great.

reinauer, 2021-03-16:
I got the dummy plugin to work by changing the code as follows:

```
 Plugin2().register()
 app = wx.App(0)
 if __name__ == '__main__':
   Plugin2().Run()

```

reinauer, 2021-03-16:
And if I do this, the InteractiveHtmlBom plugin starts fine, too:

```
diff --git a/InteractiveHtmlBom/__init__.py b/InteractiveHtmlBom/__init__.py
index 24a83a0..85cb29f 100644
--- a/InteractiveHtmlBom/__init__.py
+++ b/InteractiveHtmlBom/__init__.py
@@ -46,6 +46,8 @@ def check_for_bom_button():
 if not os.environ.get('INTERACTIVE_HTML_BOM_CLI_MODE', False):
     from .ecad.kicad import InteractiveHtmlBomPlugin
 
+    app = wx.App(0)
+
     plugin = InteractiveHtmlBomPlugin()
     plugin.register()
 

```

qu1ck, 2021-03-16:
It will start if you create wxApp in plugin but your kicad will likely not exit cleanly.

You should report this on kicad github, the issue is likely with some new version of wxpython or wxwidgets on your ubuntu 20.10

qu1ck, 2021-03-16:
Btw I just tried a fresh ubuntu 20.10 vm and can't reproduce this issue. Plugins work as expected.

Do you have some package self compiled? kicad, wxpython, wxwidgets?

reinauer, 2021-03-20:
Weird. I didn't self-compile anything but a bunch of packages have been pip-ed into the system and a mix of snap and apt packages. Might be time for a clean re-install. Thank you for your help with this. I love my BOMs now :-)

closed 2021-03-20 by reinauer

tdaniel22, 2021-05-31:
Can confirm the problem on Arch Linux with wxpython 4.1.1. Downgrading the package to 4.0.7.2 fixes the problem.

qu1ck, 2021-05-31:
You should use wxpython that kicad was compiled with. Your distro's package manager will pull the correct version. Don't install wxpython with pip.

tdaniel22, 2021-05-31:
I downgraded temporarily using Arch's official package archives, not pip: https://archive.archlinux.org/packages/p/python-wxpython/

bence98, 2023-05-03:
I have the same issue, on Linux Mint 20 and KiCad Nightly.

```
$ lsb_release -a
LSB Version:    core-11.1.0ubuntu2-noarch:printing-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID: Linuxmint
Description:    Linux Mint 20
Release:        20
Codename:       ulyana
$ py
Python 3.8.10 (default, Mar 13 2023, 10:26:41) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> wx.version()
'4.0.7 gtk3 (phoenix) wxWidgets 3.0.4'
```

```
Application: KiCad PCB Editor x86_64 on x86_64

Version: 6.99.0-unknown-68bc18425b~164~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.10.1
	HarfBuzz 2.6.4
	FontConfig 2.13.1
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux Mint 20, 64 bit, Little endian, wxGTK, plasma, x11

Build Info:
	Date: Dec 26 2022 20:33:03
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.68.0
	ngspice: 38
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON
```

qu1ck, 2023-05-03:
It's a kicad packaging issue, nothing I can do here that won't break the plugin for others.

Do other plugins with dialogs work?

You can either use the plugin from command line or try the snap version of the kicad nightly.