openscopeproject/InteractiveHtmlBom #240: No data

dotneck335, 2021-03-20:
Got it installed---put InteractiveHTMLBom in AppData/Roaming/kicad/scripting/plugins. Opened pcbnew, refreshed plugins, got icon on menu bar. Clicked on it, and got the menu for options, clicked on 'save current settings' and 'Generate BOM'. Got an Internet Explorer page with a screen that looks like a form with no entries--no data whatsoever. Warning at bottom about "restricted this website from running scripts or ActiveX controls". Clicked on 'Allow blocked content'.
NO HELP--- Still a form with no data entries.
???????

qu1ck, 2021-03-20:
https://github.com/openscopeproject/InteractiveHtmlBom/wiki/FAQ#q-generated-bom-page-doesnt-work-in-ieoperanetscapesamsung-browserkindlemy-microwave-that-can-run-doom-3d

qu1ck, 2021-03-24:
To clarify, open the generated page in a modern browser (chrome, firefox, edge, safari). IE is not supported and will not be supported.
If you have a modern browser installed but plugin still opens IE then make sure your browser is set as default to handle html files in your OS.

closed 2021-03-24 by qu1ck

dotneck335, 2021-03-25:
Thanks! That did it! How do I set my normal browser (firefox) as default to
open html files?

dotneck335, 2021-03-25:
Never mind! Found default settings and changed default browser and .html opener to Firefox,