openscopeproject/InteractiveHtmlBom #241: Filtering by hierarchical sheet

arturo182, 2021-03-26:
I'm not sure if it's possible with how KiCad files are structured but since usually whole blocks are separated into hierarchical sheets in the schematic, it would be a nice feature to be able to filter by sheets in ibom. 
This way you could assemble boards sheet by sheet and test each block as you go. It's a very specific feature and maybe it wouldn't be usable to that many, but though I'll propose it anyway on the off chance you think it's useful too ;)

qu1ck, 2021-03-26:
This will automatically work in v6 when I implement getting extra fields from pcb file as original schematic sheet is stored on footprint properties.
Then you will be able to use sheet as extra field and use standard filtering/sorting controls.

But I won't promise to backport this to 5.1

arturo182, 2021-03-27:
Awesome! And I don't mind nightly only, been living that #nightlylife for years now :)

Should I close the issue then?

qu1ck, 2021-03-27:
Nah, let's repurpose this one to track parsing extra fields from pcb file so that I don't forget.

sticilface, 2021-06-24:
I’d like to filter by hierarchical sheet as well. Is this a feature that will work with kicad 6 but not 5.1? If I understand correctly.

qu1ck, 2021-06-24:
You understand correctly. But it's not implemented yet.

commit 27a24a9f (2021-07-22): Refactor extra fields handling

ParserBase now has sensible defaults and other parsers
can implement extra field handling.

GenericJsonParser now handles extra fields in dialog mode
Kicad parser can get extra fields from pcb (5.99 only)

Fixes #241
closed 2021-07-22 by qu1ck