openscopeproject/InteractiveHtmlBom #243: Error using a standalone script on mac

LuvTian, 2021-04-20:
- environment：
MacOS
File: EasyEDA projects（pcb.json）
python version: Python 3.9.1

- Steps to reproduce and the error message:

```
➜InteractiveHtmlBom git:(master) sudo /usr/local/devenv/python/bin/pip3 install wxPython
WARNING: The directory '/Users/rehack/Library/Caches/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting wxPython
  Downloading wxPython-4.1.1-cp39-cp39-macosx_10_10_x86_64.whl (18.1 MB)
     |████████████████████████████████| 18.1 MB 93 kB/s 
Collecting pillow
  Downloading Pillow-8.2.0-cp39-cp39-macosx_10_10_x86_64.whl (2.8 MB)
     |████████████████████████████████| 2.8 MB 80 kB/s 
Collecting numpy; python_version >= "3.0"
  Downloading numpy-1.20.2-cp39-cp39-macosx_10_9_x86_64.whl (16.1 MB)
     |████████████████████████████████| 16.1 MB 149 kB/s 
Collecting six
  Downloading six-1.15.0-py2.py3-none-any.whl (10 kB)
Installing collected packages: pillow, numpy, six, wxPython
Successfully installed numpy-1.20.2 pillow-8.2.0 six-1.15.0 wxPython-4.1.1
WARNING: You are using pip version 20.2.3; however, version 21.0.1 is available.
You should consider upgrading via the '/usr/local/devenv/python/bin/python3.9 -m pip install --upgrade pip' command.
➜  InteractiveHtmlBom git:(master) which wxPython
wxPython not found
➜  InteractiveHtmlBom git:(master) python3 generate_interactive_bom.py ~/Downloads/pcb.json
```

```
This program needs access to the screen. Please run with a
Framework build of python, and only when you are logged in
on the main display of your Mac.
```

- I want to know how to fix this problem？

qu1ck, 2021-04-20:
If you [google](https://www.google.com/search?q=Please+run+with+a+Framework+build+of+python%2C+and+only+when+you+are+logged+in+on+the+main+display+of+your+Mac.&rlz=1C5CHFA_enUS864US864&oq=Please+run+with+a+Framework+build+of+python%2C+and+only+when+you+are+logged+in+on+the+main+display+of+your+Mac.&aqs=chrome..69i57.517j0j7&sourceid=chrome&ie=UTF-8) for the error message you will find multiple suggestions on how to fix the issue. Easiest one to try is running `pythonw` instead of `python`

Another thing you can try if you don't need a configuration dialog is disabling gui by defining `INTERACTIVE_HTML_BOM_NO_DISPLAY` environment variable.

LuvTian, 2021-04-21:
Use pythonw command has  an`ImportError: No module named wx` error
but I installed wxPython with pip and python3 venv
[image]

qu1ck, 2021-04-21:
You have some other pythonw on path it shows python2.7 while you installed wx in python3. Run using full path to wherever your python3 installation is.

qu1ck, 2021-05-02:
@LuvTian Was your issue resolved? I'll close this in a few days unless I hear from you.

LuvTian, 2021-05-10:
thanks ,I resolved it by INTERACTIVE_HTML_BOM_NO_DISPLAY=false