openscopeproject/InteractiveHtmlBom #244: exception on python action plugin code under kicad ver: 5.99

chopin1998, 2021-04-22:
[image]

qu1ck, 2021-04-22:
People don't read the FAQ, I guess I'll have to keep one issue about this open until v6 is released.

Use latest version of plugin from master branch if you are using nightly.

chopin1998, 2021-04-23:
sorry.... and thank thee very much!

arturo182, 2021-04-24:
@qu1ck I have notifications enabled for new issues in this repo and I admire your patience 🙏

blopblopblup, 2021-07-22:
Sorry to bring it up again, I do have an exception with 5.99 and the latest pull from master. 
[image]

tistructor, 2021-10-17:
I installed kicad's latest nightly and master plugin but I get this error window:
[image]

qu1ck, 2021-10-17:
I have no idea how that error can happen. It complains that wx has no grid attribute which is actually a submodule and it is imported in that same file. If it couldn't find the submodule it should have complained on import, not later when it's used.

Can you try opening scripting console and pasting this line by line:
```
import wx.grid
g = wx.grid.Grid()
dir(g)
```

tistructor, 2021-10-17:
['AcceleratorTable', 'AcceptsFocus', 'AcceptsFocusFromKeyboard', 'AcceptsFocusRecursively', 'AddChild', 'AddFilter', 'AddPendingEvent', 'AdjustForLayoutDirection', 'AdjustScrollbars', 'AlwaysShowScrollbars', 'AppendCols', 'AppendRows', 'AreHorzGridLinesClipped', 'AreVertGridLinesClipped', 'AssignTable', 'AssociateHandle', 'AutoLayout', 'AutoSize', 'AutoSizeColLabelSize', 'AutoSizeColumn', 'AutoSizeColumns', 'AutoSizeRow', 'AutoSizeRowLabelSize', 'AutoSizeRows', 'BackgroundColour', 'BackgroundStyle', 'BatchCount', 'BeginBatch', 'BeginRepositioningChildren', 'BestSize', 'BestVirtualSize', 'Bind', 'BlockToDeviceRect', 'Border', 'CacheBestSize', 'CalcCellsExposed', 'CalcColLabelsExposed', 'CalcGridWindowScrolledPosition', 'CalcGridWindowUnscrolledPosition', 'CalcRowLabelsExposed', 'CalcScrolledPosition', 'CalcUnscrolledPosition', 'CanAcceptFocus', 'CanAcceptFocusFromKeyboard', 'CanDragCell', 'CanDragColMove', 'CanDragColSize', 'CanDragGridColEdges', 'CanDragGridRowEdges', 'CanDragGridSize', 'CanDragRowSize', 'CanEnableCellControl', 'CanHaveAttributes', 'CanHideColumns', 'CanScroll', 'CanSetTransparent', 'CaptureMouse', 'Caret', 'CellHighlightColour', 'CellHighlightPenWidth', 'CellHighlightROPenWidth', 'CellSpan', 'CellSpan_Inside', 'CellSpan_Main', 'CellSpan_None', 'CellToGridWindow', 'CellToRect', 'Center', 'CenterOnParent', 'Centre', 'CentreOnParent', 'CharHeight', 'CharWidth', 'Children', 'ChildrenRepositioningGuard', 'ClassInfo', 'ClassName', 'ClearBackground', 'ClearGrid', 'ClearSelection', 'ClientAreaOrigin', 'ClientRect', 'ClientSize', 'ClientToScreen', 'ClientToWindowSize', 'ClipHorzGridLines', 'ClipVertGridLines', 'Close', 'ColLabelSize', 'ColLabelTextOrientation', 'ColMinimalAcceptableWidth', 'ColSizes', 'Connect', 'Constraints', 'ContainingSizer', 'ConvertDialogPointToPixels', 'ConvertDialogSizeToPixels', 'ConvertDialogToPixels', 'ConvertPixelsToDialog', 'CornerLabelTextOrientation', 'CornerLabelValue', 'Create', 'CreateGrid', 'Cursor', 'DLG_UNIT', 'DefaultAttributes', 'DefaultCellBackgroundColour', 'DefaultCellFitMode', 'DefaultCellFont', 'DefaultCellOverflow', 'DefaultCellTextColour', 'DefaultColLabelSize', 'DefaultColSize', 'DefaultEditor', 'DefaultGridLinePen', 'DefaultRenderer', 'DefaultRowLabelSize', 'DefaultRowSize', 'DeleteCols', 'DeletePendingEvents', 'DeleteRows', 'DeselectCell', 'DeselectCol', 'DeselectRow', 'Destroy', 'DestroyChildren', 'DestroyLater', 'DevicePosToGridWindow', 'Disable', 'DisableCellEditControl', 'DisableColResize', 'DisableDragColMove', 'DisableDragColSize', 'DisableDragGridSize', 'DisableDragRowSize', 'DisableFocusFromKeyboard', 'DisableHidingColumns', 'DisableKeyboardScrolling', 'DisableRowResize', 'Disconnect', 'DissociateHandle', 'DoEnable', 'DoFreeze', 'DoGetBestClientSize', 'DoGetBestSize', 'DoGetBorderSize', 'DoGetClientSize', 'DoGetPosition', 'DoGetSize', 'DoMoveWindow', 'DoPrepareDC', 'DoSetClientSize', 'DoSetSize', 'DoSetSizeHints', 'DoSetWindowVariant', 'DoThaw', 'DoUpdateWindowUI', 'DragAcceptFiles', 'DrawCellHighlight', 'DrawColLabel', 'DrawColLabels', 'DrawCornerLabel', 'DrawRowLabel', 'DrawRowLabels', 'DrawTextRectangle', 'DropTarget', 'EffectiveMinSize', 'Enable', 'EnableCellEditControl', 'EnableDragCell', 'EnableDragColMove', 'EnableDragColSize', 'EnableDragGridSize', 'EnableDragRowSize', 'EnableEditing', 'EnableGridLines', 'EnableHidingColumns', 'EnableScrolling', 'EnableTouchEvents', 'EnableVisibleFocus', 'Enabled', 'EndBatch', 'EndRepositioningChildren', 'EventHandler', 'EvtHandlerEnabled', 'ExtraStyle', 'FindFocus', 'FindWindow', 'FindWindowById', 'FindWindowByLabel', 'FindWindowByName', 'FirstFullyVisibleColumn', 'FirstFullyVisibleRow', 'Fit', 'FitInside', 'Font', 'ForceRefresh', 'ForegroundColour', 'Freeze', 'FreezeTo', 'FromDIP', 'FrozenColGridWindow', 'FrozenCornerGridWindow', 'FrozenRowGridWindow', 'GetAcceleratorTable', 'GetAccessible', 'GetAutoLayout', 'GetBackgroundColour', 'GetBackgroundStyle', 'GetBatchCount', 'GetBestHeight', 'GetBestSize', 'GetBestVirtualSize', 'GetBestWidth', 'GetBorder', 'GetCapture', 'GetCaret', 'GetCellAlignment', 'GetCellBackgroundColour', 'GetCellEditor', 'GetCellFitMode', 'GetCellFont', 'GetCellHighlightColour', 'GetCellHighlightPenWidth', 'GetCellHighlightROPenWidth', 'GetCellOverflow', 'GetCellRenderer', 'GetCellSize', 'GetCellTextColour', 'GetCellValue', 'GetCharHeight', 'GetCharWidth', 'GetChildren', 'GetClassDefaultAttributes', 'GetClassInfo', 'GetClassName', 'GetClientAreaOrigin', 'GetClientRect', 'GetClientSize', 'GetColAt', 'GetColGridLinePen', 'GetColLabelAlignment', 'GetColLabelSize', 'GetColLabelTextOrientation', 'GetColLabelValue', 'GetColLeft', 'GetColMinimalAcceptableWidth', 'GetColMinimalWidth', 'GetColPos', 'GetColRight', 'GetColSize', 'GetColSizes', 'GetConstraints', 'GetContainingSizer', 'GetContentScaleFactor', 'GetCornerLabelAlignment', 'GetCornerLabelTextOrientation', 'GetCornerLabelValue', 'GetCursor', 'GetDPI', 'GetDPIScaleFactor', 'GetDefaultAttributes', 'GetDefaultBorder', 'GetDefaultBorderForControl', 'GetDefaultCellAlignment', 'GetDefaultCellBackgroundColour', 'GetDefaultCellFitMode', 'GetDefaultCellFont', 'GetDefaultCellOverflow', 'GetDefaultCellTextColour', 'GetDefaultColLabelSize', 'GetDefaultColSize', 'GetDefaultEditor', 'GetDefaultEditorForCell', 'GetDefaultEditorForType', 'GetDefaultGridLinePen', 'GetDefaultRenderer', 'GetDefaultRendererForCell', 'GetDefaultRendererForType', 'GetDefaultRowLabelSize', 'GetDefaultRowSize', 'GetDropTarget', 'GetEffectiveMinSize', 'GetEventHan [... cut ...] owStyleFlag', 'WindowToClientSize', 'WindowVariant', 'XToCol', 'XToEdgeOfCol', 'XYToCell', 'YToEdgeOfRow', 'YToRow', '_SetTable', '__bool__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__nonzero__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__']

tistructor, 2021-10-17:
Thanks,
Will I have to wait for new builds to try the plugin?

qu1ck, 2021-10-17:
Either that or patch the diff I linked and build yourself :)

qu1ck, 2021-10-20:
Should be fixed in 25c088b but you will need to get newer kicad nightly as well.

sborshch, 2021-11-26:
KiCad 6.0.0-rc1-160-g99ec2b1801 (today's git master)
InteractiveHtmlBom cloned as git clone https://github.com/openscopeproject/InteractiveHtmlBom.git (today's master)

got error "AttributeError: 'FP_SHAPE' object has no attribute 'GetAngle'"

qu1ck, 2022-01-05:
Now that v6 is released this issue is obsolete.

closed 2022-01-05 by qu1ck