openscopeproject/InteractiveHtmlBom #248: Feature Request: Show traces of all layers in same view

mdeweerd, 2021-05-13:
First of all, thank you for this great tool.
Version: 2.3

I tested the track/net highlight function and it would be more interesting if it would be possible to show all tracks in the same view - very much like we have in the PCB drawing tool.  It would make it easier to follow a signal from one point to another even if it changes sides.

qu1ck, 2021-05-14:
This is not very useful in current representation. It only shows F and B copper layers anyway, if you have tracks in inner layers you won't be able to trace them.

Once ibom supports output templates there can be a template that renders all layers overlaid and this can be revisited.

mdeweerd, 2021-05-14:
If only Top and Bottom would be shown in the same view ("image") this is already very usefull for 2-layer boards.

qu1ck, 2021-05-16:
>  It would make it easier to follow a signal from one point to another even if it changes sides.
Point of highlight is that you don't have to follow, you see at a glance everything that is connected to the net.

Anyway, this will not be implemented in current template, it's meant to be visual representation of board as you see it in your hands. Closing in favor of #234, I'll add a note there for another potential template.

closed 2021-05-16 by qu1ck