openscopeproject/InteractiveHtmlBom #249: Feature Request: Sync zoom on Front and Back

mdeweerd, 2021-05-13:
Version: 2.3

Showing both sides of the board would be even more handy if it were possible to synchronize the zoomed viewports.  What I mean, if I soom on C23 for instance on the top, the bottom should show the same area.

qu1ck, 2021-05-14:
This will be a bit tricky when viewport sizes are different but I think it's doable.

mdeweerd, 2024-01-15:
Stale Issue, closed by the author, not a bot ;-).

closed 2024-01-15 by mdeweerd

qu1ck, 2024-01-15:
There is no harm in keeping old feature requests open. If I don't have time/motivation to implement these, maybe someone else who wants to contribute will.

But it's your issue :)

mdeweerd, 2024-01-15:
They are in my github issues backlog and I want to keep that more or less clean too.
Old issues almost never get done.