openscopeproject/InteractiveHtmlBom #250: Feature Request: Focus on the selected components

mdeweerd, 2021-05-13:
Version: 2.3

Currently, when selecting components that are not in view, the viewport do not adjust (shift) to the selected components.

I suggest:
- No click: current behavior, only highlight;
- Single click: keep the line selected to keep the component(s) hightlighted even when the mouse is no longer hovering the line in the list.
- Multiple selected components (CTRL+click to add, CTRL+SHIFT+CLICK to mark end of a list): multiple lines can be selected to highlight all components (involved);
- Double click: adjust the viewport to the components selected by the line (possibly add an option to set the maximum zoom level for this function).

qu1ck, 2021-05-14:
Auto zoom was also discussed in #246.
Multiple selection is planned as part of #206 where you will be able to "permanently" highlight some rows.

mdeweerd, 2021-05-14:
Thank your for the feedback.

It was difficult to tell from the time/initial message for the abovementioned items.  I try to keep one feature per "Issue". ;-).

mdeweerd, 2024-01-15:
Stale Issue, closed by the author, not a bot ;-).

closed 2024-01-15 by mdeweerd