openscopeproject/InteractiveHtmlBom #251: Slight misalignment of text.

phil-barrett, 2021-06-04:
This is a fantastic piece of work.

I noticed on my board that some text is not correctly aligned in the board view.  I have 2 fields that are aligned via Align/Distribute so they show up as aligned in KiCad, gerb view, OSHPark's viewer and, more importantly, on the actual PCB.  One text field is center justified and the other is left. Yet they show up with different Y values in iBOM. Not the end of the world but thought you would want to know. I can't share the Kicad files with you, sorry.

[image]
[image]

qu1ck, 2021-06-04:
I think this was fixed in 6c70a9d576a2a633b733d41643f6015329f37087

Can you try with latest ibom from master?

If the issue persists please copy paste that connector with surrounding text into an empty pcb, there should be no issues with sharing that.

phil-barrett, 2021-06-04:
Sure, is that V2.3?

qu1ck, 2021-06-04:
No, it's not in any release yet, use

[image]

direct link: https://github.com/openscopeproject/InteractiveHtmlBom/archive/refs/heads/master.zip

phil-barrett, 2021-06-04:
Yes, that also fixes it. I will close the issue.  Thanks for the quick response!

closed 2021-06-04 by phil-barrett