openscopeproject/InteractiveHtmlBom #253: Uncaught DOMException when extra fields contain spaces

cdwilson, 2021-06-13:
When I include an extra field which contains spaces in the field name, the BOM fails to load and I get the following error:

```
Uncaught DOMException: Failed to execute 'add' on 'DOMTokenList': The token provided ('Manufacturer Part Number') contains HTML space characters, which are not valid in tokens.
```

[image]

[image]

qu1ck, 2021-06-13:
Thanks for testing the dev branch and reporting bugs!

closed 2021-06-13 by qu1ck

cdwilson, 2021-06-14:
Thanks for the quick fix!

commit a9b35e49 (2021-06-13): Fix some bugs related to column ordering

Use column name as element property, not css class
Make css class predictable and not depending on extra field names

Fixes #253