openscopeproject/InteractiveHtmlBom #256: Extra Fields and HTML defaults unusable

brendanhaines, 2021-06-20:
Some, potentially all settings selected in HTML defaults tab have no effect on output. Specifically "Show Fabrication Layer" defaults to false in browser regardless of checkbox state (or contents of config.ini in plugin root).

The "Extra Fields" tab appears to have missing graphics on add/remove buttons
[image]

When an extra field is added (click top button), it is not visible in the list (no visual change at all) and I cannot enter a name
I've also tried bypassing the GUI and setting extra fields in config.ini with no more success

I also cannot select a "DNP field name". Similarly editing config.ini does not get around the problem

## Plugin version
Latest from Master  
8c6ccd2765fc3c0863208b6e8bbb74dd6deee336

## KiCad version info
```
Application: KiCad PCB Editor

Version: 5.99.0-unknown-937f5138db~130~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.8.0-55-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Jun 20 2021 08:57:00
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.3.0
	Curl: 7.68.0
	ngspice: 31
	Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

qu1ck, 2021-06-20:
> Some, potentially all settings selected in HTML defaults tab have no effect on output. Specifically "Show Fabrication Layer" defaults to false in browser regardless of checkbox state (or contents of config.ini in plugin root).

HTML settings are cached in browser the default chosen in config.ini or HTML defaults tab will have no effect if a setting is already preserved. To test if HTML default setting works open the file in an incognito browser window or use the browser's "clear cache and data" function to reset settings. If you are a web dev you can go into dev tools -> application data -> local storage and remove records there.

> The "Extra Fields" tab appears to have missing graphics on add/remove buttons

This is a wxWidgets/GTK bug. I'll see what I can do about it. But those are Up/Dn buttons, not add/remove.

> When an extra field is added (click top button), it is not visible in the list (no visual change at all) and I cannot enter a name
I've also tried bypassing the GUI and setting extra fields in config.ini with no more success
I also cannot select a "DNP field name". Similarly editing config.ini does not get around the problem

You are misunderstanding how extra fields work. You have to add them in kicad eeschema first. Then you export a netlist file or xml file generated by eeschema bom tool. Then you choose that file in Ibom (it will automatically pick the file up if it's in the project's folder) and those extra fields will appear. Then you can choose to include them in the bom or use as dnp field.

You don't add extra fields in ibom itself.

qu1ck, 2021-07-11:
Invisible button labels should be addressed in 4df0dd3

The rest is not actionable from my side so closing this. If you have questions on how to use extra fields feel free to continue discussion here.

closed 2021-07-11 by qu1ck