openscopeproject/InteractiveHtmlBom #258: Non-filled rectangles in EasyEDA dont render properly

EdHayes3, 2021-07-06:
I have a non-filled rectangle in EasyEDA.

I believe the Json for the object is:
`RECT~4452~3145~157.48~157.48~3~gge7857~0~1~none~~TEST~`

Two problems with it. 

1. The position is incorrect. Should be to the right in the big blank space. 
2. It's showing filled in. Should be just an outline.

Output:
[image]

EasyEDA Object:
[image]

qu1ck, 2021-07-07:
Thanks for the bug report. I do wish easyeda kept their docs up to date, it's not the first time they have something undocumented in their files...

closed 2021-07-07 by qu1ck

EdHayes3, 2021-07-07:
Wow! that was quick! When should I expect the published version with this fix? I'm pretty new to this github thing...

And re: undocumented items... At least it's not like Oracle.. they specifically document something in one way (functionality which makes sense, like 2+2=4), then the user finds there's a bug in the code (which is obvious, like 2+2 = 5), and they will claim it's a documentation bug. ha!

qu1ck, 2021-07-07:
I don't plan publishing a new version soon, there will be one closer to kicad 6 launch but that's ways off.

You can always get latest from master branch from this link
https://github.com/openscopeproject/InteractiveHtmlBom/archive/refs/heads/master.zip

or by going to project page, clicking code -> download zip

[image]

> At least it's not like Oracle.. they specifically document something in one way (functionality which makes sense, like 2+2=4), then the user finds there's a bug in the code (which is obvious, like 2+2 = 5), and they will claim it's a documentation bug. ha!

That's funny. Easyeda are just being lazy and haven't updated docs in years while they've been adding features.

commit aefda002 (2021-07-07): Fix easyeda RECT parsing

Also remove RECT from dataformat and json schema because
it's not actually supported in rendering.
Integrators should use segments instead.

Fixes #258