openscopeproject/InteractiveHtmlBom #259: New error with V5.99

rahuldhebri, 2021-07-11:
I have cloned the latest master branch and I am using V5.99.0-11022 when I try to run generate BOM it gives me this error "unsupported drawing class PCB_DIM_ALIGNED"
[image]

Application: KiCad PCB Editor (64-bit)

Version: (5.99.0-11022-g0527dc6fe0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.74.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jun 14 2021 08:59:27
	wxWidgets: 3.1.5 (wchar_t,STL containers)
	Boost: 1.75.0
	OCC: 7.5.0
	Curl: 7.74.0-DEV
	ngspice: 34
	Compiler: Visual C++ 1928 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

qu1ck, 2021-07-11:
Update kicad, this was fixed in https://gitlab.com/kicad/code/kicad/-/commit/fbbe72381150c1995c401f318786d2fb643113c7

rahuldhebri, 2021-07-11:
Thanks for your quick reply, Solved!

closed 2021-07-11 by rahuldhebri