openscopeproject/InteractiveHtmlBom #261: Ignone "hide" for graphic text items on silkscreen layers, KiCad

hexeguitar, 2021-07-17:
Related to:
KiCad 5.1.10
KiCad 5.99
I noticed recently that somehow all my silkscreen text items are not rendered at all, started to investigate the issue and found out something:
If you go to **Edit Text and Graphics Properties**, mark the **PCB Text Items** and uncheck the **Visible**, all gr_text items will receive flag "hide", which normally is ignored. In KiCad the text is always rendered and plotted on the gerbers. There is also no visible way to check if these items have the flag set apart from looking at the file in text editor. 
I must have accidentally marked the "PCB Text Items" while hiding all the footprint references. It was quite confusing to find out the text is not rendered in the html file, yet i can see it displayed in KiCad, in the gerbers and on the manufactured pcb i'm holding in my hand.

[image]

To solve that and make the text appear again the user has to go the "Edit Text and Graphics Properties", check the PCB Text Items, click Visible two times (uncheck->check), otherwise it means no action, leave previous setting.

So my proposal would be: to avoid confusion, maybe it will be better to ignore the "hide" flag for all gr_text items on F.SilkS and B.SilkS layers? Apparently the flag is not used in KiCad, you can't hide the silkscreen text. I have tested it on the nightly 5.99 build on Ubuntu20.4 too. Behaves the same.

qu1ck, 2021-07-17:
I just tested this both in 5.1.10 and current nightly.
Hiding text in "Edit Text and Graphics Properties" dialog DOES affect gerbers. It also DOES affect rendering in kicad unless you have "Hidden text" turned on in Appearance->Objects tab (nightly) or Items tab (5.1).

So I would say ibom does exactly what it should and reflects real state of the pcb.

If you have gerbers with refs then you generated them before you've hidden them or after you fixed them.

qu1ck, 2021-07-17:
I filed https://gitlab.com/kicad/code/kicad/-/issues/8806
Let's see how that goes before I start adding workarounds in ibom.

commit 3bd6617b (2021-07-28): Ignore visibility on PTEXT (kicad)

Fixes #261
closed 2021-07-28 by qu1ck