openscopeproject/InteractiveHtmlBom #262: Sort by component thickness

wraybowling, 2021-07-17:
Many DIYers prefer to start with the thinnest/flattest components first and work their way up to the thickest/tallest. I'm not certain that the data is readily available in libraries, but I think it would be awesome to add to iBOM

qu1ck, 2021-07-17:
Thickness is not well defined and ibom can't guess it. For some purposes tallest components are the ones that should be populated last, that is not captured in pcb in any way.
Ibom supports custom fields so you can assign your own population order to each schematic symbol, based on thickness or any other metric and use that as extra field. Then you can sort by that column in html.

wraybowling, 2021-07-18:
Custom field for now is an okay workaround. Maybe some future change to KiCAD will make thicknesses more readily available.

closed 2021-07-18 by qu1ck