openscopeproject/InteractiveHtmlBom #268: Feature request: Extra field without affecting grouping

timoscs, 2021-08-25:
It would be useful to be able to add an extra field without it affecting the grouping. 

The problem with the way it currently works:

1. make a Kicad-project which has, say, 14 100n 0603 capacitors. Add a custom field, manf# to precisely one of these, to specify the exact part number. Our production BOM export script then propagates that manf# to the other 100n caps (grouped by equal component name, footprint, and value, but crucially not by the manf#)
2. make iBom for hand-assembling the prototype, including manf# in the Extra fields. Now there is one row with 13 100n caps which have an empty manf# field, and one row which has the one remaining 100n cap, with the manf#.

Of course, we can just not add the manf# to Extra fields, but then we have to double check the actual part number from somewhere else. Most of the time, footprint + value make that possible to deduce uniquely, but in the end that still generates a small extra chance for mistakes.

qu1ck, 2021-08-25:
It's super easy to copy that manf# to all caps in eeschema in Tools -> Edit symbol fields dialog. If you copy paste value to a group it will be set on all components in the group.

qu1ck, 2021-08-29:
No problem, I'll close this in favor of #167 .

closed 2021-08-29 by qu1ck