openscopeproject/InteractiveHtmlBom #270: Feature Request: Intelligent value matching

joelsa, 2021-09-10:
Sometimes my iBOM looks a bit like this:

[image]

One reason for this is that I am trying to reuse subschematics which often were created by other people quite some time ago.
The obvious, tedious fix is to implement and enforce a project-wide naming policy and correct it for all earlier revisions.

The intelligent and definitely faster way is to have some sort of matching among different phrasings of the same value.

Similar to how https://github.com/SchrodingersGat/KiBoM supports intelligent value matching:

[image]

The implementation is not that complicated:
https://github.com/SchrodingersGat/KiBoM/blob/master/kibom/units.py

I would like to implement it, if you agree that such a feature is useful.

qu1ck, 2021-09-10:
This has been in ibom from the start pretty much

https://github.com/openscopeproject/InteractiveHtmlBom/blob/master/InteractiveHtmlBom/core/units.py#L3

But on your screenshot rows 11, 13 and 14 should have been combined so looks like there is some edge case here. Can you copy paste the table in text format or at least just those rows?

qu1ck, 2021-09-10:
Probably just variation of mu character

https://www.compart.com/en/unicode/U+03BC

joelsa, 2021-09-10:
> This has been in ibom from the start pretty much

Well, this is embarrassing. I just assumed it is not, because it did not work here,  but I should have checked the actual code instead.

Here you go:

 | Row  | Reference  | Value 
-- | -- | --
11 | C3425, C3429, C3432 | 1µ 
13 | C2306 | 1µF
14 | C2702 | 1u

So yeah, we have both https://www.compart.com/en/unicode/U+00B5 and https://www.compart.com/en/unicode/U+03BC Interestingly, the former is generated when using the [Compose Key] followed by [M] and [U] on Ubuntu, whereas the latter is the result of pressing [Alt Gr] + [M] on a German Keyboard. Only the latter is currently inside `units.py`.

joelsa, 2021-09-10:
This was fixed in KiBom recently: https://github.com/SchrodingersGat/KiBoM/commit/d9c6eb11c0663216d89cd423de716006c3f94292

qu1ck, 2021-09-12:
Fixed in 202aabff2e6085113b44a167c8e29dc31790a713

closed 2021-09-12 by qu1ck