openscopeproject/InteractiveHtmlBom #271: Documentation: install path

mxa, 2021-09-18:
On my system the installation folder is located atv`~/.local/share/kicad/5.99/scripting/plugins/` 
(kicad nightly on kubuntu 21.04)
This should be amended in the [documentation](https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Installation )

qu1ck, 2021-09-18:
AFAIK that is the default for nightly and it's based on $XDG_CONFIG_HOME env var but it's not set in all distros and can vary.
Directories currently in docs should still work.

I don't want to put it in the docs since in v6 there is a specific button to open plugins dir and also a plugin manager to make installation easy.

mxa, 2021-09-18:
Point to `Menu → Tools → External Plugins → Open Plugin Directory...` in the documentation?

qu1ck, 2021-09-18:
Done

closed 2021-09-18 by qu1ck