openscopeproject/InteractiveHtmlBom #279: Toolbar icon size

leoheck, 2021-11-17:
Hi, nice that this thing is already on PCM. 

Now, the icon looks bigger than the others

[image]

I posted the same in the forum asking if we have any guidelines for this. 
https://forum.kicad.info/t/guideline-for-plugins/32036

Gasman2014, 2022-01-26:
Think the guidelines suggest 24px x 24px - can't find the ref right now.

leoheck, 2022-01-27:
I think it was 24px and it was reduced to 22px later.

qu1ck, 2022-01-27:
It's 24x24. https://dev-docs.kicad.org/en/python/pcbnew/

leoheck, 2022-01-27:
So, if you are following guidelines with 24px it shows the guideline is clearly wrong. Look at the image above dude.

qu1ck, 2022-01-27:
I never said I am following current guideline. It used to be 32 because I wrote the first guideline. It was then reduced to 26 and now to 24.

leoheck, 2022-01-27:
Ah, cool. So the issue still stands.

leoheck, 2022-01-27:
Do you accept a PR? I can fix it if needed.

leoheck, 2022-01-27:
Here is my attempt to help.
https://github.com/openscopeproject/InteractiveHtmlBom/pull/285

commit a3293076 (2022-01-27): Resize plugin icon to 24x24 and pixel align again

Fixes #279
closed 2022-01-27 by qu1ck