openscopeproject/InteractiveHtmlBom #282: Exception with 6.99 nightly

cebrax, 2022-01-05:
Hello,

Thank you for this great plugin :)

Do you have any plans for 6.99 nightly compatibility? I am getting below exception:

[image]

qu1ck, 2022-01-05:
I have a patch for this particular issue already but wanted to wait few weeks because the API is expected to be in turmoil while some major refactors take place.

linser233, 2022-02-02:
Is there any way to use it at 6.99 now?
After getting used to ibom, it's really a big hassle not to use it.

cebrax, 2022-02-03:
Wow, seeing now all the things you are trying to do, I feel even more grateful for this plugin. Also, I think lots of other KiCad users are very grateful, too. Thank you for your wonderful efforts

sborshch, 2022-02-04:
I'm getting another exception:

  File "/opt/kicad/share/kicad/plugins/InteractiveHtmlBom/InteractiveHtmlBom/ecad/kicad.py", line 220, in parse_text
    angle = d.GetTextAngle() * 0.1
TypeError: unsupported operand type(s) for *: 'EDA_ANGLE' and 'float'

hildogjr, 2022-02-17:
My case, also Nightly 6.99:

[image]

C44Supra, 2022-03-18:
@qu1ck Any updates? I'm running into the same EDA_ANGLE issue as @hildogjr

qu1ck, 2022-03-19:
Just pushed 50c790cc6494d62d6127e38725312a0f9202ce94, it brings compatibility up to current nightly. If you are running into issues please update kicad first before reporting them, otherwise keep posting problems in this bug.

SilvanRehm, 2022-04-06:
> Just pushed [50c790c](https://github.com/openscopeproject/InteractiveHtmlBom/commit/50c790cc6494d62d6127e38725312a0f9202ce94), it brings compatibility up to current nightly. If you are running into issues please update kicad first before reporting them, otherwise keep posting problems in this bug.

Works for me now, thanks.

Updating the dcm zip in the releases would be great, as I was downloading it there first. In the wiki it is refered to as "the latest release", which made me think it's generated automatically.

In any case, thanks a lot for your work. I really takes the edge out of assembling PCBs.

qu1ck, 2022-04-06:
@SilvanRehm that wiki also has a note abot 6.99

> ...first make sure that you have latest kicad nightly and latest version of the plugin from git, not a published release.

DOGKiCad, 2022-04-27:
I've downloaded v2.4.1 and master on 6.99 and still getting Exceptios:

[image]
 
 
[image]

qu1ck, 2022-04-27:
> I've downloaded v2.4.1 and master

Don't get both, remove 2.4.1, use only master.

DOGKiCad, 2022-04-28:
I've uninstalled v2.4.1 and installed master . Still getting the same exceptions 

After extracting the ZIP files I get the below error

[image]
[image]

qu1ck, 2022-04-28:
@DOGKiCad please read the installation wiki. You can not install this plugin on kicad nightly from PCM, remove it from PCM and follow manual installation instructions from [wiki](https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Installation).

WhiteChairFromIkea, 2022-05-20:
@quick, iBOM refuses to start with current nightly (see version below). Last seen (worked on exports that day) worked v6.99 + iBOM combination was ~April 27 (that days nightly).

[image]

```
Application: KiCad (64-bit)

Version: (6.99.0-1920-g294b8e9051), release build

Libraries:
	wxWidgets 3.1.6
	FreeType 2.11.1
	HarfBuzz 4.2.0
	FontConfig 2.14.0
	libcurl/7.83.1-DEV Schannel zlib/1.2.12

Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: May 19 2022 07:19:41
	wxWidgets: 3.1.6 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.83.1-DEV
	ngspice: 36
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

WhiteChairFromIkea, 2022-05-20:
Attaching sample project:
[smpl.ZIP](https://github.com/openscopeproject/InteractiveHtmlBom/files/8737757/smpl.ZIP)

qu1ck, 2022-05-20:
Thanks for the report, this is caused by update to wxwidgets 3.1.6 which has API breaking change that wxpython has not caught up to yet. I'll see if I can word around this.

qu1ck, 2022-05-20:
Should be fixed in fdf59d8

WhiteChairFromIkea, 2022-05-20:
> Should be fixed in [fdf59d8](https://github.com/openscopeproject/InteractiveHtmlBom/commit/fdf59d8dd8fa8a061032f71f22789754ef83f80a)

Thank you! 🙂

chuckwagoncomputing, 2023-03-11:
I'm getting a similar (but different) error on Kicad 7.

```
Traceback (most recent call last):
  File "/home/runner/work/***/***/hardware/InteractiveHtmlBom/InteractiveHtmlBom/generate_interactive_bom.py", line 63, in <module>
    ibom.main(parser, config, logger)
  File "/home/runner/work/***/***/hardware/InteractiveHtmlBom/InteractiveHtmlBom/core/ibom.py", line 283, in main
    pcbdata, components = parser.parse()
  File "/home/runner/work/***/***/hardware/InteractiveHtmlBom/InteractiveHtmlBom/ecad/kicad.py", line 567, in parse
    "silkscreen": self.parse_drawings_on_layers(
  File "/home/runner/work/***/***/hardware/InteractiveHtmlBom/InteractiveHtmlBom/ecad/kicad.py", line 226, in parse_drawings_on_layers
    drawing = self.parse_drawing(d[1])
  File "/home/runner/work/***/***/hardware/InteractiveHtmlBom/InteractiveHtmlBom/ecad/kicad.py", line 191, in parse_drawing
    return self.parse_shape(d)
  File "/home/runner/work/***/***/hardware/InteractiveHtmlBom/InteractiveHtmlBom/ecad/kicad.py", line 76, in parse_shape
    a1 = round(d.GetArcAngleStart() * 0.1, 2)
TypeError: type EDA_ANGLE doesn't define __round__ method
```

qu1ck, 2023-03-11:
Update you plugin, you are running old version.

chuckwagoncomputing, 2023-03-11:
You're right; sorry for the disruption.