openscopeproject/InteractiveHtmlBom #287: trouble installing the plugin

renegadecitizen, 2022-03-15:
i have version 6 of kicad
i have installed both though the plugin manager and file system
but the plugin wont showup
can u please help me?

qu1ck, 2022-03-16:
Don't install both through plugin manager and file system. In v6 plugin manager is the recommended option.

So first, delete any files you manually copied.

Then restart kicad and check if the plugin shows up in pcb editor.

If it doesn't then go to action plugin preferences in pcb editor

[image]

If you have issues with plugin loading there will be a button with yellow warning sign at the bottom. Click it and show me a screenshot (or copy paste text) of that window.

renegadecitizen, 2022-03-16:
Thanks for helping man
It still isn’t working
Here’s a screenshot
[image]

qu1ck, 2022-03-16:
Judging by the scrollbar there's a lot of errors, please copy paste whole text.

Also what is exact os version and kicad version? Is this M1 mac? Does the scripting console open without errors? It's the right most button on the top toolbar of the pcb editor.

renegadecitizen, 2022-03-17:
https://pastebin.com/BPupiXnY

qu1ck, 2022-03-17:
This is KiCad issue, similar one was reported earlier https://gitlab.com/kicad/code/kicad/-/issues/8622#note_689417072

There is also a workaround mentioned there to increase ulimit to > 256 but I don't have a mac and can't provide exact instructions so you'll have to figure that out yourself.

renegadecitizen, 2022-03-17:
Thanks for your help

closed 2022-03-17 by qu1ck

qu1ck, 2022-03-18:
I got a response on the bug I linked above, this may also be caused with some notarization issues that should be fixed in 6.0.4. Try that release

renegadecitizen, 2022-03-18:
Ok thanks