openscopeproject/InteractiveHtmlBom #288: Accessory parts missing

marmil, 2022-03-17:
Hi everybody,
first a huge thank you for work on that project. I really like it and appreciate your time invests here.

I am encountering a small problem which i can not solve and therefore ask for your help. As it is a KiCAD standard functionality which the HTMLBom is not capable of i am assuming it is a bug.

Some of my parts are marked in the KiCAD 6.0.1 schematic as "Exclude from board" because i want to have them on my BOM but they are not placed on the PCB (e.g. Jumpers or connectors fitting to the sockets on my PCB).
[image]

I can not make the HTML BOM to show them. Is this somehow configurable and i am too stupid to find or is this not yet implemented or a bug?

qu1ck, 2022-03-17:
This is by design. Ibom is for things that are on the pcb. It gives you ability to easily copy the bom to external tool like spreadsheet and amend it there if you wish.

qu1ck, 2022-03-19:
To expand a bit: 

In general for any production there will always be parts outside of pcb that even kicad doesn't know about, stuff like housings, fasteners, crimp connectors, wiring, whatever. This bom is not designed to be a full project bom that will cover everything, there are dedicated tools for that. Goal of this project is to provide pcb documentation with emphasis on assembly assistance.

So while what you are asking for is technically possible I'm not inclined to implement this without a very good use case that fits into the mentioned goal.

closed 2022-03-19 by qu1ck