openscopeproject/InteractiveHtmlBom #291: FR: Option to dim unselected components

macourteau, 2022-04-04:
InteractiveHtmlBom is extremely useful (thanks!), and I feel like having a setting (could it be a slider?) to control the opacity of everything _but_ the highlighted components would be very useful. I can get something pretty easy to view by keeping only the "pads" layer active, but it would be nice to be able to keep other layers (in particular silkscreen and "highlight first pin") visible but dimmed.

Alternatively, having the option to make everything the same gray-ish color, except the highlighted parts, so the highlighted parts stand out (colorblind mode?) would also be useful.

qu1ck, 2022-04-04:
You can customize colors with `user.css`, see [Customization]()

Board rendering colors are defined at the top of ibom.css, you can copy this structure to your user.css and change color as you like.
```
:root {
  --pcb-edge-color: black;
  --pad-color: #878787;
  --pad-hole-color: #CCCCCC;
  --pad-color-highlight: #D04040;
  --pad-color-highlight-both: #D0D040;
  --pad-color-highlight-marked: #44a344;
  --pin1-outline-color: #ffb629;
  --pin1-outline-color-highlight: #ffb629;
  --pin1-outline-color-highlight-both: #fcbb39;
  --pin1-outline-color-highlight-marked: #fdbe41;
  --silkscreen-edge-color: #aa4;
  --silkscreen-polygon-color: #4aa;
  --silkscreen-text-color: #4aa;
  --fabrication-edge-color: #907651;
  --fabrication-polygon-color: #907651;
  --fabrication-text-color: #a27c24;
  --track-color: #def5f1;
  --track-color-highlight: #D04040;
  --zone-color: #def5f1;
  --zone-color-highlight: #d0404080;
}
```

If you want to be able to change them dynamically you should be able to do it with a bit of custom javascript.

macourteau, 2022-04-29:
Thanks for the suggestion, that's a good workaround. :)

qu1ck, 2023-08-30:
Cleaning up unactionable issues. This functionality is also related to #334, configurable stack would have configurable colors and opacity.

closed 2023-08-30 by qu1ck