openscopeproject/InteractiveHtmlBom #294: custom fields

jneiva08, 2022-04-13:
Hi,

When try to create ibom from kicad v6.x.x it doesn't appear custom fields.

symbol field table from kicad:
[image]

When try generate the ibom in the fields tab it doesn't show the custom field.
[image]

What can do to fix?

qu1ck, 2022-04-13:
How did you install the plugin? How did you launch it?
There are few odd things in your screenshots: the window shows version 6.0.0 of the plugin which doesn't exist. It also didn't load the pcb file in extra data field, it should have.

See if manually selecting extra data file (choose the same pcb file you are generating bom for) helps.

jneiva08, 2022-04-16:
I have Linux Mint 20, the kicad version is 6.0.4-6f826c9f35~116~ubuntu20.04.1, release build
I have installed the ibom by "Plugin And Content Manager"

In my work computer have the version displayed.
I have installed in my home computer and have the right version.

[image]

The image was from a project made in kicad v5, in the first time didn't show lcsc and manf#.
In the image appear because have "Update PCB with changes made to schematic" and appears.

qu1ck, 2022-04-16:
In KiCad v5 to import extra fields you had to export netlist file and then ibom would pick it up. In KiCad v6 it's enough to just update pcb from the schematic, like you said, ibom can pull fields from v6 pcb file.

closed 2022-04-16 by qu1ck