openscopeproject/InteractiveHtmlBom #301: CLI mode depends on folder name

qu1ck, 2022-05-25:
Prior to 234c744ce56f0faef3d328216ea76f8cdd7f49fa the code used to be agnostic to what the folder name of the InteractiveHtmlBom was exactly.

Now because of that change the CLI mode doesn't work when the plugin is installed from PCM because it official repository requirements dictate it to be org_openscopeproject_InteractiveHtmlBom (see [this report](https://forum.kicad.info/t/interactive-html-bom-plugin-for-kicad-5-0/11713/325) ).

@set-soft I don't really understand what issue was solved by that change? Is there another workaround that would make the script work in your use case but also not be dependent on exact folder name?

set-soft, 2022-05-26:
Ok, the first option seems to be the cleanest.

I tried:

```
.
└── org_openscopeproject_InteractiveHtmlBom
    ├── __init__.py
    ├── InteractiveHtmlBom
    │   ├── generate_interactive_bom.py
    ....

```

This is just the layout of the repo. The `__init__.py` file is the one at the root of the repo, which imports the plugin. All the relevant files are inside `InteractiveHtmlBom` (including the `generate_interactive_bom.py` script).

In this way the name of the package remains `InteractiveHtmlBom` and the KiCad directory is `org_openscopeproject_InteractiveHtmlBom`. The `org_openscopeproject_InteractiveHtmlBom/__init__.py` is the bridge between both, and this is the file you have in the repo.

set-soft, 2022-05-26:
BTW: The `generate_interactive_bom.py` file should be executable.

qu1ck, 2022-05-26:
> When you install it at system level this isn't true, the CLI command goes to /usr/bin/ and the module to /usr/lib/python3/dist-packages/. This is for a regular Linux system. Even if you install it using pip and the user HOME you'll get something like: CLI command goes to ~/.local/bin and the package to ~/.local/lib/python3.9/site-packages/ (or similar, depending on the Python version).

That's where I assumed that a symlink would go to bin folder and the script itself would remain in the package and continue to work as is because it reads real path. If this is not considered kosher by debian/setuptools standards then I'd rather do this:
1. Move any actual logic out of generate_interactive_bom.py into `__main__.py`
2. Leave generate_interactive_bom.py as a dumb launcher for legacy reasons, ideally making it execute `__main__.py`
3. Exclude generate_interactive_bom.py from any package installers, they should generate their own wrappers if needed

Reason why there is another `__init__.py` on one level above InteractiveHtmlBom in the repo is solely to make the plugin easier to install for people who just checkout the git repo into their kicad plugins folder. The files outside of InteractiveHtmlBom are not part of the package, just source files that shouldn't be distributed to end users.

I don't like the java-style package naming forced upon us in the official PCM repo either but it serves a purpose.

set-soft, 2022-06-03:
Ok, PR #306 does something similar:

- I reverted the code to use relative imports and fake the package name, so it works locally the same way it did before.
- To allow global installation I moved around some code in `generate_interactive_bom.py` and added an entry point called `main()`. So now `setup.py` just registers the entry point and the installation creates a `/usr/bin/generate_interactive_bom.py` that just calls the entry point.
- I also protected the plugin registration from `generate_interactive_bom.py`. This is because the wrapper created by easy install initializes the package (to find the entry point) before we even have a chance to define the environment variable. I didn't remove the environment variable, just added an extra condition.

I tried running it locally, as a plugin and installed at system level. All worked.

PR #306 (2022-06-07): Use relative imports in generate_interactive_bom.py
- Also avoid registering as plugin if we are:
  `generate_interactive_bom.py`
- Created a main() entry point (used by setup.py)

Fixes #301
closed 2022-06-07 by qu1ck