openscopeproject/InteractiveHtmlBom #302: Add a way to import DNP / blacklist afterward

lukh, 2022-05-30:
Hi ! 
I was wondering if it could be possible to add a way to import / inject a blacklist after the file generation...
(context: i have a two steps build process to generate CAD files:
- One for gerber, ibom, PNP files, that are not fab dependant (build with kicad, InteractiveHTmlBom, and the kicad_pcb file)
- Another one that uses customs scripts on BOM (csv, .pos, etc)  to define what is PNP, what is mounted manually, etc
This second step is completely independant, and should not (can not) relate on kicad_pcb file

-> i'd like to add a menu entry in the ibom file, to load a csv/txt or a list of references. It updates the BOM as following:
If the entry is in the list, then it check a "DNP" / "Already Mounted" checkbox on the BOM list...

qu1ck, 2022-05-30:
> define what is PNP, what is mounted manually, etc
This second step is completely independant, and should not (can not) relate on kicad_pcb file

Do you still have the kicad file at that point? If you do, the easiest way to achieve what you want is to generate new ibom file through CLI and pass blacklist in arguments.

If you really need to be able to filter out things from the ibom opened in browser you can do it with a bit of custom javascript, see Customization section in wiki.
You would need to remove entries from `pcbdata.bom` tables (see DATAFORMAT) and then call `populateBomTable()`

lukh, 2022-05-31:
I technically not have it at that point (it 's git tracked, and the pack (gerbers, pos, ibom, schematics) for manufacturing and doc is generated via gitlab CI (kiauto)
The idea is to decorrelate the need for kicad when producing.
Thanks, I start to get used to the code

closed 2023-04-29 by qu1ck