openscopeproject/InteractiveHtmlBom #308: Add %o option for output directory

hildogjr, 2022-06-06:
Add "%o" option to save the BOM into output directory used for Gerbers and other files.

Using version 2.4.1 on Kicad 6.99

qu1ck, 2022-06-06:
> output directory used for Gerbers and other files.

Do you mean the last used path that pcbnew remembers in the plot dialog? It's not available in api, I think it's in project json file.

I imagine this would only be useful if you have non-standard output directory per project or generate multiple variants of the outputs or something like that. There are external projects that take care of this kind of automation (kibot), I don't think this is something Ibom should handle. 

Or am I misunderstanding the request?

hildogjr, 2022-06-06:
This, this is correct (the dialog path).
Usually I use "output - project" as this path, but the sometimes the "project title" is to long and I abbreviate it for the output folder name. So the path became different of "output - %p".

hildogjr, 2022-06-06:
The information is available on `(setup (pcbplotparams (outputdirectory "???")))` pcb file but I think it is not exposed through the Python API.

hildogjr, 2022-06-06:
I could be it!
```python
import pcbnew
output_folder = pcbnew.GetBoard().GetPlotOptions().GetOutputDirectory()
```

hildogjr, 2022-06-06:
Implemented on https://github.com/openscopeproject/InteractiveHtmlBom/pull/309

qu1ck, 2023-08-30:
This is kicad specific and not broadly useful. I'm not opposed to merging a fix for this but you'll have to put the work in and fix the issues I mentioned on your MR.

closed 2023-08-30 by qu1ck