openscopeproject/InteractiveHtmlBom #310: [help] Class for change userheader.html

hildogjr, 2022-06-06:
@qu1ck, this is just a help need.

In the past you explained me how to correct included an embedded SVG icon on top of the generated HTML file for iBOM.

```html
<div style="float: left; margin: 6px; background-color:white;">
<svg width="176" height="54" viewBox="0 0 178 56">
...
</svg>
</div>

Now I want to modify this icon (invert the color or change the trace color when I active the dark mode. Which CSS could I use to detect this change? For example, some variable of the class that returns black/white depending of of the visualization mode.

qu1ck, 2022-06-07:
You can't do this with just css.

The way dark mode works in ibom is simply by adding `dark` class to `#topmostdiv` element.
Then any element that needs it's colors tweaked for dark theme has special `.dark .something` rule in css.
It's done this way to make it easier to integrate the ibom page into other websites the way for example kitspace did it, it doesn't mess up original website design.

Because your logo element is outside of the `#topmostdiv` this will not work out of the box.

Here's few things you can do:
1. Move your logo element inside `#topmostdiv`. You will have to dynamically inject your logo using js instead of just putting it inside html header. Then you can invert colors using css by adding `.dark .mylogo` rule to user css file.
2. Override the `setDarkMode()` function in user js to call the original but also do any additional styling you want.

qu1ck, 2022-09-19:
Closing for inactivity, reopen if you need further help.

closed 2022-09-19 by qu1ck