openscopeproject/InteractiveHtmlBom #311: Does not work in KiCAD 6.99

spflanze, 2022-06-07:
I attempted to run this plugin in KiCAD 6.99 in Pop!_OS 20.04 and got a long list of errors. The command sequence I used to install this plugin is:

`mkdir ~/.kicad/

mkdir ~/.kicad/scripting/
mkdir ~/.kicad/scripting/plugins
cd ~/.kicad/scripting/plugins
git clone https://github.com/openscopeproject/InteractiveHtmlBom.git`

Did I install the correct version of InteractiveHtmlBom?

The errors:
[image]
.

qu1ck, 2022-06-07:
Yes. Did you remove the plugin from plugin manager? You should.

qu1ck, 2022-06-10:
If you still have issues lets continue in #282

closed 2022-06-10 by qu1ck