openscopeproject/InteractiveHtmlBom #317: Help

the555timer, 2022-07-17:
[image]

qu1ck, 2022-07-17:
You are likely using CLI mode. This is kicad issue, press cancel and it will work fine.

the555timer, 2022-07-18:
[image]

qu1ck, 2022-07-19:
Did you change how you are running the plugin? Those error messages don't seem compatible.

In the last screenshot you are using the bat file, likely for EasyEDA, in that case you need to install wxpython in your system python environment. Run this in command line
`pip install wxpython`
and try again.

the555timer, 2022-07-19:
I already have wxpython installed. I also tried reinstalling it, but no luck. I even tried to run it using anaconda becauose usualy that solves many of the library problems.

qu1ck, 2022-07-19:
The error straight up says in white on black, no wxpython.

This can happen if you have multiple python environments and the one that is used to run the plugin (if you are running from bat file it's the default one that is on PATH) is not the same one where you installed wxpython.

Open cmd and tell me what the output of these commands is:
`python -version`
`python -m pip show wxpython`

the555timer, 2022-07-19:
`python -version` is `python -V` for me, but it returns `Python 3.9.7`.
`python -m pip show wxpython` returns

 `Name: wxPython`
`Version: 4.1.1`
`Summary: Cross platform GUI toolkit for Python, "Phoenix" version`
`Home-page: http://wxPython.org/`
`Author: Robin Dunn`
`Author-email: robin@alldunn.com`
`License: wxWindows Library License (https://opensource.org/licenses/wxwindows.php)`
`Location: c:\users\márk\appdata\local\programs\python\python39\lib\site-packages`
`Requires: numpy, pillow, six`
`Required-by:`

qu1ck, 2022-07-19:
Then try to run plugin in the same cmd window directly without batch file
`python .../generate_interactive_bom.py .../board.json`

the555timer, 2022-07-20:
[image]

the555timer, 2022-07-26:
I did not edit the code but I changed it to wx but I still get an error 
[image]

qu1ck, 2022-07-26:
Looking at your output, you can ignore these debug assertions. It seems to me that everything worked, check for the html file in the `bom` directory on your desktop (where board file is).

But still, if you didn't edit the code then someone else did. You should get rid of your modified plugin and download fresh from github to be sure there are no more surprises.

the555timer, 2022-07-26:
I can't find the html file, I looked for it on my desktop and in the folder where the pcb scource file is located also there is no bom folder on my desktop

qu1ck, 2022-07-26:
Ok, I was able to replicate this issue with official python and wx distribution. Somehow python that comes with kicad doesn't have the same problem.

You can work around this for now by calling plugin directly in command line as in [this comment](https://github.com/openscopeproject/InteractiveHtmlBom/issues/317#issuecomment-1188890804) without batch file or by removing the `--show-dialog` argument in the batch file.

qu1ck, 2022-07-26:
This should be fixed now. Download latest from master.

commit 49f64121 (2022-07-26): Create wx app as first step before any imports

Fixes #317
closed 2022-07-26 by qu1ck