openscopeproject/InteractiveHtmlBom #318: DNP must be in extra_fields

MeisterBob, 2022-07-20:
For filtering out DNP components, the DNP field must be in extra_fields as well. That results in an empty DNP column in the BOM (because all components with non empty DNP field gets filtered out). Is this behavior required by some circumstance that I can't see or can we remove
https://github.com/openscopeproject/InteractiveHtmlBom/blob/2f38736d8506f45c6a0945eccea1497e02fef1af/InteractiveHtmlBom/core/ibom.py#L71
and replace
https://github.com/openscopeproject/InteractiveHtmlBom/blob/2f38736d8506f45c6a0945eccea1497e02fef1af/InteractiveHtmlBom/core/ibom.py#L72
with a check that DNP is 1 or True and not 0 or False?

qu1ck, 2022-07-20:
> For filtering out DNP components, the DNP field must be in extra_fields as well

Yes but it is not required to be shown. I just tested dnp field, I can set it and not tick "show/group" and it works as expected. Are you experiencing something different?

qu1ck, 2022-07-20:
Also all extra fields are strings so DNP is considered True for any non-empty string, even if its "false" or "0".

MeisterBob, 2022-07-20:
I use brd2json to create the json file from eagle. Then I run the json through a python script that searches for values starting with "nb" and adds `"DNP": "true"` to the `extra_fields` (the script also adds our internal Article No and storage location but that doesn't matter here).
Now comes the iBom magic. I generate the bom with
```
.../generate_interactive_bom.py board_name.json --highlight-pin1 --bom-view left-right --layer-view F --name-format board_name --extra-fields "ArtNo" --dnp-field DNP
```
but the DNP components are still in the bom. If I generate the iBom with
```
.../generate_interactive_bom.py board_name.json --highlight-pin1 --bom-view left-right --layer-view F --name-format board_name --extra-fields "ArtNo,DNP" --dnp-field DNP
```
the DNP components disappear but I get an empty DNP column.

MeisterBob, 2022-07-20:
I digged into the Code a little deeper. Unselected extra_fields are removed in `GenericJsonParser.parse()`, so you can't check the dnp_field in `skip_component()` if it's not in the extra_fields list. I created a pull request with a possible fix for that.

PR #319 (2022-07-21): allow dnp without dnp field in extra_fields
fixes #318
commit f9e4ceb1 (2022-07-21): allow dnp without dnp field in extra_fields (#319)

* allow dnp without dnp field in extra_fields

fixes #318

* genericjson: add board_variant_field to extra_fields
closed 2022-07-21 by qu1ck