openscopeproject/InteractiveHtmlBom #320: Coordinate origins

hindenbugbite, 2022-07-20:
I'm working on adding IPC2581 as a new ecad format. I'm in a bit of a mental loop about the coordinate origin being on the upper left as stated in the dataformat page. I'm imagining that if I feed it coordinates based on lower left origin that it will draw the board vertically flipped (mirrored?), so does this mean the code must discover and translate all coordinates so that:

1) if y values are all negative then flip the sign then assume origin is upper left?
2) if y values are positive then have to subtract it from board height to get the value from upper left?

I'm assuming that the board origin is arbitrary from board to board and negative values are okay. Just this idea of rendering a mirrored board has me spinning cycles on how to determine origin and translating. If I'm way off base, that would be great.

qu1ck, 2022-07-20:
Negative values are ok. As a result all you have to do is multiple all y coordinates by -1. If you want them to be positive then translate them all by adding a constant.
So you get `Ynew = -Yold + C` where `C` can be `max[Yold]`.

qu1ck, 2022-08-15:
Feel free to reopen if you have more questions. I'd like to see what you got whenever you are ready to show it off.

closed 2022-08-15 by qu1ck