openscopeproject/InteractiveHtmlBom #321: [CLI][Discussion] frightening and unimportant messages displayed to the user

set-soft, 2022-08-12:
The following code:

https://github.com/openscopeproject/InteractiveHtmlBom/blob/49f6412128751f71853f685d3e013bde47f83b65/InteractiveHtmlBom/version.py#L20-L22

Displays some debug messages that I don't think are relevant to the end user. Why the code assumes the script is running from a repo clone? 
I think these messages should be displayed only when debugging (BTW I don't think the verbosity can be controlled from the CLI).

My main current problem is that when I try to find the iBoM version I must filter these messages. They could also be printed to stderr, along with other messages created by KiCad, but most people will get confused anyways.

What do you think about this @qu1ck ?

qu1ck, 2022-08-13:
Code doesn't assume it, it just tries to get more exact version info with the commit sha for debug purposes.
Print statements can be reworded to indicate this is just informational or removed altogether.

set-soft, 2022-08-16:
I think they should be removed, after all this isn't an error, just the code running normally as installed from the KiCad package manager. Do you want a PR removing them (I think commenting them is the best, so you can uncomment them for debug).

commit 31b20f84 (2022-08-16): Remove print statements about git version check

fixes #321
closed 2022-08-16 by qu1ck