openscopeproject/InteractiveHtmlBom #323: `Description` field missing on Linux but present in Windows

eeintech, 2022-09-08:
## Windows 10

`Description` field is present:

[image]

## Linux (Ubuntu 22.04)

`Description` field is **missing**:

[image]

qu1ck, 2022-09-08:
Description field shows up only if netlist is used as extra data file. Ibom automatically picks the latest modified file as source of extra data and on your linux machine it probably happens to be the pcb file and not netlist file.

eeintech, 2022-09-09:
Thanks, we found out what was happening:
- In Linux, you were right, your plugin did not pick explicitly the `.net` file because it was a bit older than the `.kicad_pcb` file. After pointing to the `.net` file, the `Description` field appears.
- On Windows, my colleague had generated a BOM using the `KiBOM` tool which also generates a `.xml` file from the schematics, and it appears that your plugin also picks `.xml` files as source of extra fields which is why the `Description` field was showing even though we did not explicitly export the netlist.

Mystery solved :smile: Thanks again!

closed 2022-09-09 by eeintech