openscopeproject/InteractiveHtmlBom #326: Suggest to add components position (x, y, rot) in the table

waiweng83, 2022-09-13:
Suggest to have an option to include components position information (x, y, rot) in the table.
This combines BOM and position file into a single file and very useful when sending to third party for assembly process.

qu1ck, 2022-09-13:
It doesn't make much sense to me to combine BOM (grouped components table) with per-component data which is position info.
How do you envision it?

waiweng83, 2022-09-13:
> It doesn't make much sense to me to combine BOM (grouped components table) with per-component data which is position info. How do you envision it?

In this case, the components will not be grouped together.
We have a problem where we put custom fields in the components library (MPN, Manufacturer, Internal SKU, etc). These information only get exported on the BOM but not the position file. Assembly house will need to combine both BOM file and position file manually.

qu1ck, 2022-09-13:
So you don't need a bom. You just need a component list with both position data and custom fields.
I am not inclined to add this because it would need to pull additional data from pcb like aux origin location and some settings. And it's not relevant to main purpose of this plugin.

What you should do is write a plugin that will generate pos file and then add extra fields in a post processing step. You can use [KiBot](https://github.com/INTI-CMNB/KiBot) to automate these steps. Actually that project fits well for your use case in general, I suggest you open a feature request for this there.

waiweng83, 2022-09-13:
Ok, got it.

Thanks.

closed 2022-09-13 by waiweng83