openscopeproject/InteractiveHtmlBom #331: Wishlist: netlist on all layers available please, for awesome visualisation!

MATTERealise, 2022-10-12:
I have been using this tool with huge gratitude for a year or so now as I assemble prototypes. However, I just stumbled on the netlist export and view options and this would make a killer tool for communicating to people who don't have EDA software installed / on their phone/tablet(!) about where signals run. 

Is there *any* chance you could add the option to:
- export all copper layer traces
- display all copper layer traces on the front/back view simultaneously for the selected net
- perhaps even show different layer traces with different colours to highlight where they have changed layer in the view requested in the bullet point above

If you could do the first two of the above, that would make it a kick-ass portable visualiser for designs. Thanks for your amazing work.

qu1ck, 2022-10-12:
A number of web viewers for kicad files exist that do what you want. For example https://eyrie.io/
I don't have plans to have more copper layers displayed as it doesn't help with the main use case of this tool: assembly help.

MATTERealise, 2022-10-12:
Fully understand that stance and thank you for pointing me to an
alternative. If you ever change your mind, please voice it loudly as I'll
be jumping to update the plugin!

closed 2022-10-12 by MATTERealise