openscopeproject/InteractiveHtmlBom #335: Dockerfile

Granjow, 2022-11-16:
Hi guys,

thanks for this wonderful plugin.

Unfortunately it is currently a pain to get running on Ubuntu (Python 3.10+ compatibility missing) and an even bigger pain to get it running in a CI pipeline without X server, so I created a Docker image for that:

https://hub.docker.com/r/granjow/kicad-interactive-html-bom

If you are interested, I can add it to the project in a pull request.

Thanks,
Simon

qu1ck, 2022-11-16:
About X server see https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Tips-and-Tricks

And what's wrong with python 3.10 compatibility?

Granjow, 2022-11-17:
Oh, so somebody else has run into the issue already :) The docker image is not quite documented though and no sources available. I would add the Dockerfile to this project so everybody can modify it according to their needs. (Also based on Python 3 and KiCad 6.0, the solution in the wiki probably would not process 6.0 files anymore.)

wxPython did not build here with Python 3.11 as described in https://github.com/wxWidgets/Phoenix/issues/2296, but I checked again and it was because I still had `attrdict` installed after installing `attrdict3`. Without `attrdict` it builds fine.

qu1ck, 2022-11-17:
I can add your dockerfile to wiki but I don't think it should be in the tree itself.
Also you chose the hard way with xvfb, if you use the environment variable your image will be much smaller.

Granjow, 2023-02-04:
Can you add a link to it? The repo is here:
https://git.granjow.net/escape/rnd/docker-interactive-bom

Thanks!

qu1ck, 2023-02-06:
Done

closed 2023-02-06 by qu1ck

set-soft, 2023-02-06:
BTW: [KiBot](https://github.com/INTI-CMNB/KiBot/) docker images (and GitHub action) contains iBoM (I think since 2020).