openscopeproject/InteractiveHtmlBom #336: Hide DNP val/ref

Unichord, 2022-11-22:
I tried to tinker around and look through old issues but I can't seem to find this feature.
I'd like to have an option to hide text(s) only of components which are not going to be mounted, because of variants/DNP. It's already fine as it is, but I think it would be clearer to not even have unneccessary values on the board.
Maybe it can be done more easily by hiding them directly from KiCad (temporarily)?
For my use case, only values can be hidden, but maybe also hiding references can be useful to someone?

Also, the possibility to differentiate the colors of SMD/THT could be useful to many I believe. I don't know if that's feasible.

Sorry if these are already been asked, and thanks for all your work.

Unichord, 2022-11-22:
Well, I planned to leave References visible, which are physically on the board even if they are DNP, so I agree with you on that. Values, on the other hand, are not.
Even if the DNP components are never highlighted when hovering over the component list, the Value of DNP components _on the board_ can be misleading.
I believe a switch to hide Values _which are not actually on the board_ due to DNP/variants would be useful.

qu1ck, 2022-11-22:
I think it's not values that should be hidden but all of fabrication layer for footprints that are dnp.

Problem is with current architecture it's very unlikely to be implemented because graphics are processed before dnp components are determined.

Maybe in v3 or when/if the data format changes.

Unichord, 2022-11-22:
That's a bummer, I'll wait patiently.

What about the different color for SMD and THT?

qu1ck, 2022-11-23:
There are already 6 different colors that pads can be rendered and not a lot of color space left to differentiate stuff in 2 themes and 3 highlight modes. I suggest you use the checkbox and mark when checked features to highlight whichever components you want.

Unichord, 2022-11-23:
I see, well it's not a big deal anyway, just a suggestion. I was more interested in the hidden Value for DNPs. Thanks.

For anyone interested, as a workaround I'm using a `priority` field so as to sort THT components last.

qu1ck, 2025-10-24:
@jacoblikesblue parts are automatically excluded from bom if they are marked DNP in kicad or if you add a custom field that you then chose as dnp field in ibom settings, there is an explicit option for that.

Maybe you mean that DNP components still show up on the drawing in ibom and that is intentional. Drawing is supposed to represent real world board and the pads or silkscreen does not disappear because you don't intend to solder a part. 

Exception to that is fab layer because it's not physical anyway. I have plans to optionally hide it for dnp parts but it needs a data format change. I think there is a way to do it in a backwards compatible way though so stay tuned.