openscopeproject/InteractiveHtmlBom #341: Resistor color codes

Guustaaf, 2022-12-23:
First: the Interactive HTML BOM project is awesome and amazing!

How about: if you hover the cursor over a resistor the color code appears at the top?

Guustaaf

qu1ck, 2022-12-23:
You mean the axial resistor stripe colors? Not a bad idea but those parts are on their way to a museum. 

Still, this can be done with a custom column + some user javascript (see customization page in the wiki).

Guustaaf, 2022-12-26:
Through hole parts still have their place, mainly in the prototyping phase. I suppose it could be added to the value column, as well as what the value looks like on SMD parts. Same for the three digit code on capacitors. Great project!