openscopeproject/InteractiveHtmlBom #346: sometimes footprint highlighted to the frame boundary

sborshch, 2023-01-10:
There is lot of same footprints in pcb, but only some of them looks weird.
After any new BOM regeneration random number of random footprints looks weird.

PCB to reproduce: [pcb.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/10380606/pcb.zip)

cmdline to generate:

`python3 <path_to_kicad>/share/kicad/plugins/InteractiveHtmlBom/InteractiveHtmlBom/generate_interactive_bom.py --hide-silkscreen --checkboxes  --bom-view top-bottom --layer-view F --dest-dir . --name-format %f_bom --sort-order DD,DA,U,VT,VD,A,R,C,L,FB,~,ZQ,FU,JMP,JP,XP,XS --show-fabrication --include-tracks --netlist-file bomus_hv_power.net bomus_hv_power.kicad_pcb
`

Result:

[image]

InreactiveHtmlBom version:
```
<path_to_kicad>/kicad/share/kicad/plugins/InteractiveHtmlBom$ git log -1
commit ae27915f08a3f06ba4dc0a8360a181b96c442a8a (HEAD -> master, origin/master, origin/HEAD)

```
KiCad version:

```
Application: KiCad PCB Editor

Version: (7.0.0-rc1-70-g9298defd3f-dirty), debug build

Libraries:
	wxWidgets 3.0.5
	FreeType 2.11.1
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.13

Platform: Linux Mint 21.1, 64 bit, Little endian, wxGTK, cinnamon, x11

Build Info:
	Date: Jan 10 2023 09:00:53
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.1
	Curl: 7.81.0
	ngspice: 31
	Compiler: GCC 11.3.0 with C++ ABI 1016

Build settings:
	KICAD_SPICE=ON
	KICAD_STDLIB_DEBUG=OFF
	KICAD_STDLIB_LIGHT_DEBUG=OFF
	KICAD_SANITIZE_ADDRESS=OFF
	KICAD_SANITIZE_THREADS=OFF

```

qu1ck, 2023-01-10:
Kicad api is returning random stuff. I filed this issue: https://gitlab.com/kicad/code/kicad/-/issues/13465

qu1ck, 2023-01-12:
Should be fixed in next nightly.

sborshch, 2023-01-12:
Unfortunately, no. Version: (7.0.0-rc1-84-g481c2493a4), release build still has this issue.

qu1ck, 2023-01-12:
It doesn't include the fix, wait for next one.

sborshch, 2023-01-12:
It was today's git HEAD, so it include fix for https://gitlab.com/kicad/code/kicad/-/issues/13465

qu1ck, 2023-01-12:
Your version is commit 481c2493
Look at the log https://gitlab.com/kicad/code/kicad/-/commits/master it's ~20 commits behind the fix.

sborshch, 2023-01-12:
Ops.. sorry, you are right. Something wrong was with my build environment, clean build solved the problem. Thanks.

qu1ck, 2023-01-12:
Thanks for the bug report.

closed 2023-01-12 by qu1ck