openscopeproject/InteractiveHtmlBom #349: First pin highlight inconsisten (e.g. too **fat**)

eflukx, 2023-01-16:
When selecting "Highlight first pin", the markings can become way too **fat** on several (most) components. Some components are properly highlighted, but most are not. Also when there are for example a couple of the _same_ components are in a row, some may have the '_fat_' highlight, while others are correctly highlighted.

See example screenshot:
- Q8 is ok, Q7 not
- R31 is ok, R29 not
- etc.

[image]

qu1ck, 2023-01-16:
What's your kicad version and plugin version?

Can you share the design or a minimal/stripped version that shows the issue?

qu1ck, 2023-01-16:
I was able to reproduce it, fix committed.

closed 2023-01-16 by qu1ck

eflukx, 2023-01-17:
That's mighty quick indeed my friend! 😉 

Sorry I forgot the version nrs. indeed, for the record here they are:

 - Kicad 6.0.10 (release) on Windows 
 - Plugin version v2.5.0 (installed from the KiCad repo)

commit d02d0abd (2023-01-16): Reset stroke width after footprint drawings

Fixes #349