openscopeproject/InteractiveHtmlBom #350: Cannot get kicad 5.1.12 to recognize this plugin

mmagliaro, 2023-01-22:
This is on a Win 7  machine.
Downloaded the .zip, put the InteractiveHtmlBom folder into C:\Program Files\KiCad\share\kicad\scripting\plugins, which I have done for other plugins (and it worked).  But I do not see InteractiveHtmlBom in the list of plugins in pcbnew
(restarted KiCAD, and also tried "refresh plugins".

qu1ck, 2023-01-22:
Try the latest, I just pushed a fix for py2 compatibility.

mmagliaro, 2023-01-23:
That fixed it!  Awesome tool, by the way.
THANK YOU

qu1ck, 2023-01-23:
No problem.

Keep in mind, next version of ibom will probably be last to support python 2 and therefore kicad 5.1 on windows.

You should upgrade to kicad 7 soon anyways :) It has a lot of new features and is much more polished.

closed 2023-01-23 by qu1ck