openscopeproject/InteractiveHtmlBom #351: IndexError: list index out of range

jdt1, 2023-01-27:
On running the Python script on a recent .brd file, I get an IndexError on line 788.

```python
 package = \
                [pac for pac in library.find('packages').findall('package')
                 if pac.attrib['name'] == el.attrib['package']][0]
```

The list is empty, hence the index 0 is out of range. 

```python
[pac for pac in library.find('packages').findall('package')]
```
returns an element with name "CAPC1608X85"

However `el.attrib['package']` returns 'CAPC2012X110'.

I believe this is the root of the problem, but I understand what is causing it.

I can share the .brd file if needed.

Jim

Funkenjaeger, 2023-01-27:
Having the .brd file would be helpful to run this down - please share.  From what you describe, it sounds like there's a component referencing a library & package where the library does exist under `<libraries>` but the referenced package doesn't exist within that library.  That doesn't sound good, but there may be a subtlety I'm missing so I'd like to see the actual file in question.

jdt1, 2023-01-27:
Forgot to mention, the error occurs in fusion_eagly.py, but I guess that was probably clear. The file is for a client project so I cannot share it publicly, but I can share it with you in private. Can I email it to you?

Funkenjaeger, 2023-01-27:
Of course, I don't know a way to pass my email address to you privately on github either, so I set up a burner.  Shoot me an email  ibom _at_ funkenjaeger.net and I'll get back to you with my real address to send the file to.

jdt1, 2023-01-27:
You've got mail

Funkenjaeger, 2023-01-27:
Hmm, didn't come through for some reason.  Since you have your website linked in your github profile, I emailed you at the contact address listed on your site.

jdt1, 2023-01-28:
Made a typo in the email address, sorry. I sent it again, hopefully it works now!

Funkenjaeger, 2023-01-31:
Got your .brd file, and fortunately it was an easy fix.  The problem was that the .BRD file contains two libraries identically named "Capacitor" which I hadn't anticipated.  Created PR #352

jdt1, 2023-01-31:
That does seem to fix the problem, awesome!

Kind regards,

Jim

commit b0da3ecf (2023-02-03): Improve .BRD library package lookup (fix #351) (#352)

* Handle multiple libraries with duplicate names (#351)

* Eliminate f-strings for kicad 5.1 compatibility
closed 2023-02-03 by qu1ck