openscopeproject/InteractiveHtmlBom #355: wxwidgets debug alert

Hcreak, 2023-02-14:
python:  Python 3.9.13
wxpython:  4.2.0

[image]
[image]
[image]

qu1ck, 2023-02-14:
Is this happening on latest from master?

Hcreak, 2023-02-14:
> Is this happening on latest from master?

latest [Releases v2.5.0](https://github.com/openscopeproject/InteractiveHtmlBom/releases/tag/v2.5.0)

qu1ck, 2023-02-14:
Download latest from github master branch, not release.

Hcreak, 2023-02-14:
> Download latest from github master branch, not release.

Thx. this is true.
With the [Releases v2.5.0](https://github.com/openscopeproject/InteractiveHtmlBom/releases/tag/v2.5.0), I suggest annotating or deleting it.  so as not to cause the same error.

closed 2023-02-14 by Hcreak