openscopeproject/InteractiveHtmlBom #356: Installation failure

BeeniGit, 2023-02-15:
Hello,

I have tried many times to install the plugin on my kicad (6.0.10). I have also tried to reinstall kicad with-out success...
When I try to install the plugin, I got this error message : 'https://github.com/openscopeproject/InteractiveHtmlBom/releases/download/v2.5.0/InteractiveHtmlBom_v2.5.0_pcm.zip' Failure when receiving data from the peer

I also tried to install with the download zip of the repo but this time the error message is : No Json File found.

I don't know how can I solve this issue...

Thank's for your help

qu1ck, 2023-02-16:
> Failure when receiving data from the peer

That's a known issue with 6.0.10, update to 6.0.11 or better yet 7.0

Alternatively:

> I also tried to install with the download zip of the repo

download zip file from the link in your post, that one is suitable to install from plugin manager.

BeeniGit, 2023-02-18:
Thank's  for your help.

I have update with the 6.0.11 version. Everything works fine !

closed 2023-02-18 by BeeniGit