openscopeproject/InteractiveHtmlBom #357: Oversized Footprint Outline

myamigo, 2023-02-16:
I have a small FPGA in my design that has a properly defined footprint but the red outline in the ibom.html output is extending way past the courtyard. Part is U18 below...

[image]

After closer examination it seems the plugin is using the dimension line objects on the User.Comments layer to determine the extents for the part. I added a yellow box around the dimension lines after doing a screen grab from the Footprint Editor...
[image]

The red and yellow frames are perfectly correlated. If the courtyard is defined then that might be the best object to use to determine the red box dimensions. If there is no courtyard then use the Fab layer objects and barring that perhaps use the silk and pads. In any case I wouldn't use the User.* layer objects for this purpose.

Great plugin! 👍

qu1ck, 2023-02-16:
Ibom doesn't look at any specific layer contents, it uses bounding box logic of kicad. If kicad thinks those drawings are within the box, ibom will draw it truthfully.

I agree that dimension objects should not be included when text is not included. But that should be fixed on KiCad side.

myamigo, 2023-02-16:
I see. KiCad doesn't also consider the associated text so there's some inconsistency there for sure. It's a minor thing so moving on.

qu1ck, 2023-04-29:
Fixed in kicad with https://gitlab.com/kicad/code/kicad/-/merge_requests/1522. This should be in 7.0.2

closed 2023-04-29 by qu1ck